Renaud11232/OctoPrint-Resource-Monitor #1: Suggestion: Select/Filter what is displayed

ShagoY, 2019-12-04:
It would be nice to be able to filter the display, for example remove the network lo and/or Lan when I use only wifi (wlan).

Renaud11232, 2019-12-08:
Hello,

I'll look into adding this when I have some spare time.
I'm still not shure what would the best/cleanest way of doing it, but I didn't really had the time to think about this.

Renaud11232, 2019-12-30:
Hello,

Just to give an update.
I've done some modifications to the plugin that adds a way to select which network intefaces and which disks to filter out.
This is not release yet and needs further testing.

tmagis, 2020-01-06:
[image]

commit 56483b45 (2019-12-30): added a way to hide disks
fixes #1
closed 2020-01-06 by Renaud11232