Renaud11232/OctoPrint-Resource-Monitor #2: Feature request: Temperature

mittkonto, 2019-12-05:
Love this plugin!
Would be awesome though if it could monitor temperature as well!

"cat /sys/devices/virtual/thermal/thermal_zone0/temp"
Gives: 57000 = 57c

skohls, 2019-12-08:
I second this request.

Some other options:

pi@octopi:~ $ cat /sys/class/thermal/thermal_zone*/temp
40894

pi@octopi:~ $ vcgencmd measure_temp
temp=39.0'C

pi@octopi:~ $ /opt/vc/bin/vcgencmd measure_temp | tr -d "temp=" | tr -d "'C"
40.0

All taken within a few seconds - so there are some differences between them but for the purpose of this plugin they all should be ok as a data source.

Renaud11232, 2019-12-08:
Hello,

I will probably add this when I have the time.
Thanks for giving different ways of implementing this.

While these methods will likely work fine, they are linux specific, or even Raspberry specific and I want the plugin to be compatible with any OS.

`psutil` (the library I use to get resource usage) does provide a way of getting temperature sensors, that's probably how I'll end up doing it.

Thanks anyway for the advice :)

Renaud11232, 2020-01-02:
Hello,

Quick little update on this.
I've added the CPU temperature in the CPU tab.
I still need to test it on a Raspberry Pi before releasing it.
[image]

skohls, 2020-01-03:
Thanks, really looking forward to this!

tmagis, 2020-01-06:
[image]

commit 1aef4343 (2020-01-02): added cpu temperature in CPU tab
fixes #2
closed 2020-01-06 by Renaud11232