Renaud11232/OctoPrint-Resource-Monitor #3: Temperature graph

ilyko96, 2020-02-17:
It is probably a good idea to have a screen with cpu temperature graph (for raspberry pi it is `vcgencmd measure_temp` command).

This might be helpful to debug parameters (e.g. temperature thresholds) of active cooling for RPi. For example in my setup RPi is controlled by GPIO and I'm not 100% sure if my parameters are optimal (i.e. produce least noise with highest cooling performance). Having this graph could help me finetune cooling setup.

Renaud11232, 2020-02-18:
Hello,
While waiting for the graph to be added, the temperature should already be monitored in the Processor section.
[image]

ilyko96, 2020-02-18:
Yes, sure, thank you

Renaud11232, 2020-02-24:
This is released in 0.2.1

closed 2020-02-24 by Renaud11232