Renaud11232/OctoPrint-Resource-Monitor #7: Feature request: Historical data over longer period

kind3r, 2020-05-14:
Hi, 

I think it would be nice to have access to aggregated data (1min, 5 min) over a longer period most useful for longer (usually overnight) prints in case something went wrong and you want to check if:
- it was a bottleneck on the PI's CPU
- or temperature too high
- maybe it ran out of disk space because of your timelapse settings

and corelate that with the moment of your fail.

Renaud11232, 2020-05-17:
Hello,

If I understand correctly, you'd like something like a setting to set the length of the period during which the data are shown (which is 1minute at the moment) ?

I can see this being quite useful :)

I'll give it a go when I have some time

kind3r, 2020-05-17:
Indeed, but it does not need to be up to the second like it is now for the 1min interval, it can be aggregated so you don't need to store all the values. One average per minute is more than enough.
I think this should be an opt-in feature. I'm running on a PI 4 with 4G and a ssd stick so ram/disk is not an issue but it might be for others running on SD Card for example.

JasonShallcross, 2021-01-11:
Just to add, I'd definitely love to be able to see some more historical data - ideally updating dynamically.
That could be just a dropdown of "Last minute", "Last hour", "Last 24 hours" or something.
Or the ability to change the interval, "Update every minute".

My Pi was crashing, I think due to overheating. I've since bought a fan, but it would have been great to see if it had overheated around the time it stopped working.

Maybe we could set an alert point, where it then adds to a log file?

commit 88a5064f (2021-01-23): graphs are now initialized with the right configuration
fixes #16
fixes #7
closed 2021-02-07 by Renaud11232