Renaud11232/OctoPrint-Resource-Monitor #9: Print stops with error in middle of print job or just before staritng print job

theChef613, 2020-06-04:
Hello,
After the print job preheats, the print stops and the following error occurs:
`(Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2823)`

sometimes the happens immediately, sometimes it happens after a while. Uninstalling Resource monitor solved the issue.

Thank you!

Renaud11232, 2020-06-05:
Hello,

Can you provide logs and tell me what environment are you running the plugin on ? (OctoPrint version, Raspberry, ...)

I'll see if i can reproduce the issue

theChef613, 2020-06-05:
Thanks for the response! I've attached the log file below. I've attached the log file from the failed print. I'm using Octoprint 1.4.0 on a Raspberry PI 3B+ for a Prusa MK3S. Full disclosure, I reinstalled resource monitor and did a short print to try to recreate the issue but it did not happen. One time it took hours for the failure to occur, other times it happened immediately. I think this problem occurred on both dates of the log files attached.
[octoprint.log.2020-06-03.txt](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/4737366/octoprint.log.2020-06-03.txt)

[octoprint.log.2020-05-31.txt](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/4737335/octoprint.log.2020-05-31.txt)

theChef613, 2020-06-08:
Hi. was testing it out to see if I was doing something odd on my side and it happened again. I'm attaching another log file here.
[octoprint.log.txt](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/4743717/octoprint.log.txt)

vr6syncro, 2020-06-17:
Had the same after installing. Seems Ressource Monitor use serial port or blocked it.

Renaud11232, 2020-06-17:
Hello,

I haven't been able to reproduce the issue on my machine.
So far all test prints I've been doing have succeeded.

The plugin should interract with the serial port.
If `psutil` (the library I use to retrieve resource usage) does, that will need a lot of work to get fixed.

I'll keep investigating.

vr6syncro, 2020-06-19:
I use Klipper as Firmware. Klipper seems to use Serial Port. My Printer with Marlin use USB Port.
Maybe thats why u could not reproduce it?

Deapper, 2022-06-01:
This also caused prints to stop for me