Renaud11232/OctoPrint-Resource-Monitor #10: Added DS18B20 1-wire sensor via Enclosure plugin and Resource-Monitor stopped showing temperature

ShaneF1, 2020-06-23:
This is not a big issue for me but thought you'd like to know.

Prior to adding the Enclosure Plugin (https://github.com/vitormhenrique/OctoPrint-Enclosure) and adding a DS18B20 1-wire sensor to GPIO4 as described, Resource-Monitor showed a temperature reading on a Pi4 with OctoPrint 1.4.0 running on OctoPi 0.17.0.

After adding the DS18B20 sensor, the Resource-Monitor plugin no longer shows the CPU temperature. It doesn't even appear in the legend on the graph. The reading still exists and can be read at /sys/devices/virtual/thermal/thermal_zone0/temp.

[image]

Renaud11232, 2020-07-01:
Hello,

First of all, sorry for the late answer, I really don't have a lot of time lately.

Regarding the issue, I'll look into it when I have some time.

I may get back to you later if I have news or need more information.

Renaud11232, 2020-07-11:
Hello,

In order to view if it's an issue with my plugin or with the library I use to collect resource usage, can you run the following command on your Raspberry Pi, and give me what it outputs.
```bash
/home/pi/OctoPrint/venv/bin/python -c "import psutil; print(psutil.sensors_temperatures())"
```
(Assuming OctoPrint is installed in `/home/pi/OctoPrint`)

Thanks

Renaud11232, 2020-08-13:
Closed in #15

closed 2020-08-13 by Renaud11232