Renaud11232/OctoPrint-Resource-Monitor #14: ResourceMonitor hogging a whole CPU-Core on the Client

lryb, 2020-07-18:
I got a new RaspberryPi 4 2GB yesterday and installed a clean OctoPi Image from their website. I updated Octoprint to 1.4.0 and installed this plugin (no other 3rd party plugins installed). After restarting OctoPrint and reloading suddenly my ui ran very laggy. Opening Taskmanager shows me, that the browser is hogging a whole cpu-core (as much as the JS of a single tab can get). This doesn't stop after some minutes. Deactivating this plugin, restarting OctoPrint and reloading the ui solves the issue. 

This happens in the newest Brave and Firefox. It doesn't matter if connected to the Pi via Wifi or Cable. No errors in the browsers consoles.

lryb, 2020-07-20:
Hi again. I have good news. I tried around some more and realized the plugin isn't the problem. The problem was a misbehaving userscript included via tampermonkey. I had it installed in Firefox that I used previously and in brave, that I'm using currently. After deactivating the userscript Octoprint in general and especially with this plugin enabled runs way better.

I should have tried everything with a fresh browser install. My bad. Sorry for the fuss.

closed 2020-07-20 by lryb

Renaud11232, 2020-07-20:
Hello,

I'm happy to see you got it working.
Thanks for letting me know :)