Renaud11232/OctoPrint-Resource-Monitor #15: Raspberry Pi SoC Temperature no longer present

ChaosBlades, 2020-07-26:
I believe this is related.

https://github.com/imrahil/OctoPrint-NavbarTemp/issues/74

https://github.com/imrahil/OctoPrint-NavbarTemp/pull/78
"sys.platform on Python 3.3 and above no longer includes a version number. That makes the check for SoC support fail. This fix resolves that issue."

bvrielink, 2020-07-31:
This plugin doesn't use sys.platform in the code. Also I use Python 2.7 and it doesn't  show the temperature part as well. Don't know the exact problem though. If I run the temperature part on the command-line it returns results:

source /home/pi/oprint/bin/activate
python -c "import psutil; print psutil.sensors_temperatures()"
{'cpu_thermal': [shwtemp(label='', current=57.996, high=None, critical=None)]}

Renaud11232, 2020-07-31:
Hello,

I think I found the issue, thanks for providing the output from `psutil`.
I think this is related to #10.

I am currently not able to test this myself, but I just created a new pre-release [0.2.5rc1](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/releases/tag/0.2.5rc1) that should fix the issue.

You can try it and tell me if it solved the problem.

bvrielink, 2020-07-31:
> Hello,
>
> I think I found the issue, thanks for providing the output from `psutil`.
> I think this is related to #10.
>
> I am currently not able to test this myself, but I just created a new pre-release [...]

Quickly fixed. 👍 Changed to underscore and  it works!

ChaosBlades, 2020-08-13:
Got tired of waiting on this being added to master so I tried the pre-release. Works for me.

commit e668375a (2020-07-31): fixes temperature not showing
fixes #15
closed 2020-08-13 by Renaud11232