Renaud11232/OctoPrint-Resource-Monitor #18: Temperature in Fahrenheit

natron734, 2020-09-11:
Would it be possible to show Temperature tab in F?

Renaud11232, 2020-10-01:
Hello,

I just created a new release candidate that adds Fahrenheit conversion ([0.2.6rc2](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/releases/tag/0.2.6rc2))
It's not available through OctoPrint's automatic updates yet, but you can install it manually if you want to test it.

If you do so, can you report if that worked well ?

Thanks

natron734, 2020-10-02:
In installed 0.2.6rc2 and the Resource Monitor tab is no longer displayed. Note: I am using Python 3, and I disabled Themeify Plugin as additional testing. Reinstalled 0.2.6rc2 and still no Resource Monitor tab is displayed.

Renaud11232, 2020-10-02:
Can you provide the logs so I can look into the issue ?

natron734, 2020-10-02:
[plugin_pluginmanager_console.log](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/5319326/plugin_pluginmanager_console.log)

natron734, 2020-10-02:
I hope that's the log you're looking for. Thanks again!

Renaud11232, 2020-10-03:
Can you also provide `octoprint.log` ?

Thanks

natron734, 2020-10-03:
[octoprint.log](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/5321573/octoprint.log)

Renaud11232, 2020-10-08:
Hello,

Thanks for the logs, I think I found the issue.
This should now be fixed in [0.2.6rc3](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/releases/tag/0.2.6rc3)

commit bb9e409e (2020-09-14): adds Fahrenheit support
fixes #18
closed 2020-11-02 by Renaud11232