Renaud11232/OctoPrint-Resource-Monitor #19: Plugin not showing in Octorprint

Blue-RR, 2020-09-28:
Hi,

I've installed the plugin and there was no errors. However, the plugin does not show in Octorprint settings.

My setup is as follows:
System - Raspberry Pi 3B
Octorprint - 1.4.2
Python - 3.7.3

Octoprint log file show the below error. I have removed and re-installed with no luck.

`2020-09-28 10:00:23,023 - octoprint.server.views - ERROR - Error while retrieving template data for plugin resource_monitor, ignoring it
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/server/views.py", line 681, in fetch_template_data
    vars = implementation.get_template_vars()
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_resource_monitor/__init__.py", line 75, in get_template_vars
    cpu=self.get_cpu(),
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_resource_monitor/__init__.py", line 63, in get_cpu
    frequency=psutil.cpu_freq()._asdict(),
AttributeError: 'NoneType' object has no attribute '_asdict'
2020-09-28 10:00:23,037 - octoprint.server.views - ERROR - Error while retrieving template data for plugin resource_monitor, ignoring it
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/server/views.py", line 681, in fetch_template_data
    vars = implementation.get_template_vars()
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_resource_monitor/__init__.py", line 75, in get_template_vars
    cpu=self.get_cpu(),
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_resource_monitor/__init__.py", line 63, in get_cpu
    frequency=psutil.cpu_freq()._asdict(),
AttributeError: 'NoneType' object has no attribute '_asdict'`

Renaud11232, 2020-10-01:
Hello,

I just published a new release candidate that adresses this issue ([0.2.6rc2](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/releases/tag/0.2.6rc2)).
You can install it manually through OctoPrint's plugin manager.

If you do so, can you report if that fixed the issue ?

Blue-RR, 2020-10-02:
> Hello,
> 
> I just published a new release candidate that adresses this issue ([0.2.6rc2](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/releases/tag/0.2.6rc2)).
> You can install it manually through OctoPrint's plugin manager.
> 
> If you do so, can you report if that fixed the issue ?

Hi,

Just did the update and the issue has been fixed. The plugin shows up in the dashboard and in settings now.

Thanks looking into it so quickly.

Cheers,

commit 9cde6bb6 (2020-10-01): cheks if cpu frequency is None
fixes #19
closed 2020-11-02 by Renaud11232