Renaud11232/OctoPrint-Resource-Monitor #20: Fan Control (Feature Request)

Kiendeleo, 2020-11-16:
I am curious if you would be willing to add the ability to control a PWM fan via a couple of GPIO header pins.  Your plugin already polls the CPU temp, so I would think it would not be hard to add the ability to control a fan.  

There is a forum post [here ](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=244194&start=25#p1511085)on the RPi forums that already has a script written to control the fan and a wiring diagram for the PWM Control.  

If your interested, please let me know, I'd be happy to help.

cp2004, 2021-02-14:
FYI, the Raspberry Pi has built in support for PWM fans using a `dtoverlay`, or configured using `raspi-config`.

Here's a post that explains how this can be done:

https://community.octoprint.org/t/can-octoprint-control-the-new-raspberry-pi-case-fan-on-off-temperature/28284/2?u=charlie_powell

Renaud11232, 2022-02-06:
Hello,

Since this plugin aims only at monitoring the resources and this can be done directly by configuring the RPi, i'll close this issue

closed 2022-02-06 by Renaud11232