Renaud11232/OctoPrint-Resource-Monitor #22: Plugin installed but not appearing

smn86, 2020-12-13:
OctoPrint is installed on the following setup:
PCEngines APU1C4 with dual core G-T40E cpu and 4Gb of ram
SD disk with F2FS filesystem
Ubuntu 20.04.1 x86_64

The Octoprint installation works properly but, even if the resource-monitor plugin is correctly installed it doesn't show up on the plugin list.
The plugin manager however shows it installed.

The same plugin is also installed on an arm board with the same versioning and works properly.
Are there any additional system packages that needs to be installed? Is it an incopatibility with the APU Board?
Regards
Simone

Renaud11232, 2020-12-13:
It looks like there is a bug.

Can you provide the full logs ? Can you also send the output of the following commands : 

```bash
/path/to/your/venv/bin/python -c "import psutil; print(psutil.net_io_counters(pernic=True))"
/path/to/your/venv/bin/python -c "import psutil; print(psutil.net_if_addrs())"
/path/to/your/venv/bin/python -c "import psutil; print(psutil.net_if_stats())"
```

Where `/path/to/your/venv` is the directory where you installed octoprint.

Thanks

smn86, 2020-12-13:
Dear Renaud11232,
I wouldn't bet that this is a bug, the following is the output you've requested:

for - ./OctoPrint/venv/bin/python -c "import psutil; print(psutil.net_io_counters(pernic=True))"

----------------------------------------------
{'lo': snetio(bytes_sent=30793, bytes_recv=30793, packets_sent=384, packets_recv=384, errin=0, errout=0, dropin=0, dropout=0), 'enp1s0': snetio(bytes_sent=0, bytes_recv=0, packets_sent=0, packets_recv=0, errin=0, errout=0, dropin=0, dropout=0), 'enp2s0': snetio(bytes_sent=0, bytes_recv=0, packets_sent=0, packets_recv=0, errin=0, errout=0, dropin=0, dropout=0), 'enp3s0': snetio(bytes_sent=0, bytes_recv=0, packets_sent=0, packets_recv=0, errin=0, errout=0, dropin=0, dropout=0), 'wlp4s0': snetio(bytes_sent=25419033, bytes_recv=10172275, packets_sent=26412, packets_recv=20667, errin=0, errout=0, dropin=0, dropout=0), 'simnet': snetio(bytes_sent=580814, bytes_recv=398731, packets_sent=3312, packets_recv=2585, errin=0, errout=0, dropin=0, dropout=0)}
----------------------------------------------

for - ./OctoPrint/venv/bin/python -c "import psutil; print(psutil.net_if_addrs())"

----------------------------------------------
{'lo': [snicaddr(family=<AddressFamily.AF_INET: 2>, address='127.0.0.1', netmask='255.0.0.0', broadcast=None, ptp=None), snicaddr(family=<AddressFamily.AF_INET6: 10>, address='::1', netmask='ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff', broadcast=None, ptp=None), snicaddr(family=<AddressFamily.AF_PACKET: 17>, address='00:00:00:00:00:00', netmask=None, broadcast=None, ptp=None)], 'wlp4s0': [snicaddr(family=<AddressFamily.AF_INET: 2>, address='172.16.0.15', netmask='255.255.255.192', broadcast='172.16.0.63', ptp=None), snicaddr(family=<AddressFamily.AF_INET6: 10>, address='fe80::714d:2e78:4aec:f0fc%wlp4s0', netmask='ffff:ffff:ffff:ffff::', broadcast=None, ptp=None), snicaddr(family=<AddressFamily.AF_PACKET: 17>, address='00:24:2b:71:ff:49', netmask=None, broadcast='ff:ff:ff:ff:ff:ff', ptp=None)], 'simnet': [snicaddr(family=<AddressFamily.AF_INET: 2>, address='10.254.200.53', netmask='255.255.255.0', broadcast=None, ptp='10.254.200.53'), snicaddr(family=<AddressFamily.AF_INET6: 10>, address='fe80::30de:6eb1:ee8b:d4c9%simnet', netmask='ffff:ffff:ffff:ffff::', broadcast=None, ptp=None)], 'enp1s0': [snicaddr(family=<AddressFamily.AF_PACKET: 17>, address='00:0d:b9:37:e4:a8', netmask=None, broadcast='ff:ff:ff:ff:ff:ff', ptp=None)], 'enp2s0': [snicaddr(family=<AddressFamily.AF_PACKET: 17>, address='00:0d:b9:37:e4:a9', netmask=None, broadcast='ff:ff:ff:ff:ff:ff', ptp=None)], 'enp3s0': [snicaddr(family=<AddressFamily.AF_PACKET: 17>, address='00:0d:b9:37:e4:aa', netmask=None, broadcast='ff:ff:ff:ff:ff:ff', ptp=None)]}
----------------------------------------------

for - ./OctoPrint/venv/bin/python -c "import psutil; print(psutil.net_if_stats())"

----------------------------------------------
{'lo': snicstats(isup=True, duplex=<NicDuplex.NIC_DUPLEX_UNKNOWN: 0>, speed=0, mtu=65536), 'enp1s0': snicstats(isup=False, duplex=<NicDuplex.NIC_DUPLEX_UNKNOWN: 0>, speed=65535, mtu=1500), 'wlp4s0': snicstats(isup=True, duplex=<NicDuplex.NIC_DUPLEX_UNKNOWN: 0>, speed=0, mtu=1500), 'simnet': snicstats(isup=True, duplex=<NicDuplex.NIC_DUPLEX_FULL: 2>, speed=10, mtu=1500)}
----------------------------------------------

[octoprint.log](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/5685047/octoprint.log)

Just for information, disk structure is as follows:
sda1-> boot [ext4,read only]
sda2-> system root (overlayroot) [ext4,writable overlay on RAM tmpfs]
sda3-> Octoprint partition, [F2FS, writable]

Networking:
The machine is reachable through local LAN via 172.16.0.15 and through VPN via 10.254.200.53.
Let me know whether you need more information.
Regards
Simone

smn86, 2020-12-13:
Could it be some missing packages/libraries needed to check hardware status?

Renaud11232, 2020-12-14:
Hello,

I don't think it is a dependency issue, since the only library I use to retrieve hardware status is `psutil` which seems to be installed correctly, since all 3 didn't throw any error.

From the output of the 3 commands, it seems that the stats (which include link speed) for the NIC `enp2s0` are not returned by `psutil`. I can't do much about that.

But what I can do, is preventing my plugin to unexpectedly crash if this happens. I just published a pre-release that should include a fix for this. ([0.2.7rc1](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/releases/tag/0.2.7rc1)).

Can you tell me if that fixed the issue ?

Thanks

smn86, 2020-12-14:
Hello,
I've just tested the 0.2.7rc1 and it works, the Resource Monitor correctly shows up in the plugin manager and Octoprint interface.
I might just try to put the interface in the exceptions list to see whether the 0.2.6 works.

commit a7956f2d (2020-12-14): adds check for missing network stats
fixes #22
closed 2020-12-17 by Renaud11232