Renaud11232/OctoPrint-Resource-Monitor #25: [Bug] Exclude non-physical file-system types from the list

Sophist-UK, 2021-01-09:
The list of file systems on my Ubuntu based OP system includes bind and snapfs (?) mounts.

We need to have a list of fstypes that should be excluded in the statement at line 86:
```
partitions = [partition._asdict() for partition in psutil.disk_partitions() if partition.fstype and (all or partition.mountpoint not in self.__disk_exceptions)]
```
probably by changing it to something like:
```
partitions = [partition._asdict() for partition in psutil.disk_partitions() if partition.fstype and (all or partition.mountpoint not in self.__disk_exceptions) and partition.fstype not in ["bind", "snapfs"]]
```
I will double check the file system types tomorrow and try a manual edit on my system and then submit a PR.

Sophist-UK, 2021-01-09:
Ok - a little experimentation shows that you can test for fstype="squashfs" for snap mounts, but that a bind mount looks like any other mount and there is nothing to distinguish the root mount for the partition from a bind mount. (In other words I have an ext4 file system mounted at /mnt/data and bind mounts for /mnt/data/var as /var etc. but there is nothing in the psutil.disk_partitions response (fstype, device, opts or maxpath) to distinguish /var from /mnt/data.

Since for me it is snap's squashfs mounts that are frequently changing, I will submit a PR to eliminate these.

closed 2021-01-10 by Renaud11232

Renaud11232, 2021-01-10:
Thanks for the PR, it's now merged and released

PR #26 (2021-01-10): Eliminate squashfs file systems disk list
Fixes #25
commit 8888cb7c (2021-01-10): Eliminate squashfs file systems disk list

Fixes #25