Renaud11232/OctoPrint-Resource-Monitor #29: no data is shown in charts

jkhyper, 2021-02-08:
since I've updated the plugin to the latest 0.3.0 release, no data is shown in the graphs or elsewhere on teh Resoure Monitor tab.
[image]

Renaud11232, 2021-02-08:
Hello,

Could you please provide the octoprint logs as well as the browser console logs (On Google Chrome, press F12)

Thanks

mmatesic01, 2021-02-08:
I can also confirm that with version 0.3.0. Previous version worked fine.

mmatesic01, 2021-02-08:
I've just re-enabled my Resource monitor again and for some strange reason it works now. 
Could it be the bowser cache that I cleared lately?

Renaud11232, 2021-02-08:
It is possible but I can't be 100% sure

jghake, 2021-02-09:
I'm having the same issue. I cleared local files in Chrome but I haven't tried disabling the plug-in yet.

Kenjekenny, 2021-02-10:
Had the same issue after updating to version 0.3.0. After updating another plugin (DisplayLayerProgress) and rebooting octoprint, resource monitor suddenly started working again (previous reboots had no effect).

farbefreak, 2021-02-12:
I do have the same problem. After the Update there is no data displayed. 
I read this thread, so I disabled the plugin, restarted, enabled, restarted -> problem gone.
However after another restart, the data is again not displayed. -> Browser data clearing does not help. 
Only disabling and reenabeling does work.

Renaud11232, 2021-02-12:
I do understand that this seems to be an issue affecting multiple people.
But, as I said, in previous messages, I need the logs from octoprint and the JS console in order to start investigating.

The issue does not seem to happen on my OctoPrint instance, so without those logs I currently have no idea what the issue might be.

Thanks

farbefreak, 2021-02-12:
i am happy to provide logs, however i need to know what you need. i am a beginner, so i don't know what JS console means. Are you referring to something inside the browser or do i get it out of the PI?

Renaud11232, 2021-02-12:
Everything should be available from the browser.

The logs from OctoPrint can be retrieved from the settings:
[image]

To get the logs from the JS console, From Google Chrome you can press F12, there should be a "Console" tab.
You can copy paste or take a screenshot of what's there.
[image]

I'm not sure the F12 shortcut works on other browsers but you can also get there by right-clicking anywhere and selecting "Inspect element" then going in the console tab.

Thanks

jgugler, 2021-02-12:
I'm having the same issue since a couple of days. Here are the logs and the console
[octoprint.log](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/5974415/octoprint.log)
[image]

Renaud11232, 2021-02-12:
Thanks,

I didn't find any relevant error in `octoprint.log`. The logs from the browser however showed an error from my plugin.

I just create a prerelease that should address this issue, it won't automatically update but you can install it manually through the plugin manager using this url : [https://github.com/Renaud11232/OctoPrint-Resource-Monitor/archive/0.3.1rc1.zip](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/archive/0.3.1rc1.zip)

Can you test it and tell if that fixed the issue ?

Thanks

jgugler, 2021-02-12:
Just installed it and after restarting it works again! Thanks

Renaud11232, 2021-02-13:
Thanks
@jkhyper @mmatesic01 @jghake @Kenjekenny @farbefreak
Can you also check that it's fixed for you as well ?

Thanks

jghake, 2021-02-13:
Looks like mines working

Kenjekenny, 2021-02-13:
Resource monitor stopped working again after reboot but after installing your bugfix its working again, even after numerous reboots. Thx alot!

farbefreak, 2021-02-13:
I did install it and its working!

Many many thanks for your work!

closed 2021-02-13 by Renaud11232

Renaud11232, 2021-02-13:
The fix for this issue is now released in version 0.3.1 that should be available directly in OctoPrint.

jkhyper, 2021-02-13:
> The fix for this issue is now released in version 0.3.1 that should be available directly in OctoPrint.

Thank you for the fast followup and support. With 0.3.1 all data is shown again. Also thanks for maintaining a useful plugin!

commit 7a26fa37 (2021-02-13): release 0.3.1
fixes #29