Renaud11232/OctoPrint-Resource-Monitor #34: Resource monitor not showing any data.

whpgab, 2021-11-01:
Hi! 
I update my octoprint to the latest. And Resource monitor is 3.5. Now whatever I doing it not showing any data on charts. Strange things when I restart octo and showing Reload button on the background there is some datas on charts. 

[image]

Renaud11232, 2021-11-01:
Hello,

Can you please send the logs from octoprint with the plugin logs set to "DEBUG" ?

Thanks

whpgab, 2021-11-01:
I hope that is good for you.
Thanks

[octoprint (9).log](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/7455741/octoprint.9.log)

Renaud11232, 2021-11-04:
Hello,

I've setup an OctoPrint server with the exact same plugins as you and I can't seem to reproduce the issue.

Can you please copy/paste the results when you go on the following URL : `<octoprint_ip>/plugin/resource_monitor/stats`
(replace `<octoprint_ip>` by the real IP adresse of OctoPrint)
You'll need to be authentified for it to work.

Can you also copy/paste the contents of 1 message from the plugin.
To get this (using Google Chrome) : 
1. Press F12 to open developper tools
2. Go in the "Network" tab
3. Search for `websocket` and click on it
4. Open the "Messages" tab
5. Search for "resource_monitor"
6. Right click on one of the last messages in the list (it will refresh every second)
7. Copy the message
8. Paste it here

Not knowing how confortable you are with the Chrome Dev tools, here is an image explaining how to do it :
[image]

Thanks

whpgab, 2021-11-05:
Dear Renaud11232,

I don't know what happend, but since then it is working. There were two updates for Octoprint. From 1.7.0 to 1.7.1 and  1.7.2. Maybe that was the problem. Thanks for your support and help!

Renaud11232, 2021-11-06:
Hello,

Since you got it working, I'll close this issue.

closed 2021-11-06 by Renaud11232