Renaud11232/OctoPrint-Resource-Monitor #37: [BUG] Print Job Intermittently Paused Then Continues

sarf2k4, 2022-04-14:
During print job, with a lot of g-code command that has to be sent to the printer, printer intermittently pauses a bit and then continues the print job. Leaving a small blob on where it pauses. This has been confirmed with after I removed a lot of other plugins I suspected of such problems but the issue still persists. Octoprint in safe mode didn't have any of this issues

Before this problem arise, I did `update` and `upgrade` command on my octopi to keep it updated

**To Reproduce**

1. Enable this plugin with stock settings of 1 seconds interval (I think).
2. Any gcode that has intricate or very minimal movement.

**Expected behavior**

This plugin shouldn't interfere or has higher priority than sending commands over serial

**Version of OctoPrint-Resource-Monitor**

0.3.6

**Server configuration**
[image]
[image]

**OctoPrint logs with `octoprint.plugins.resource_monitor` set to DEBUG**

I am not sure if this intermittent pauses will appear on the logs

Renaud11232, 2022-04-14:
Hello,

The logs will probably not show the pauses but they might show errors or other issues that might help me figuring out what the issue is.

Can you confirm that this issue does happen only when this plugin is installed ? And/or stopped happening once it was disabled/uninstalled ?

As of now, I really don't know what could cause this or have any way to find out since I use the plugin on my printer and have not had this issue so far.

One possible cause is that the CPU of the Pi is 100% loaded but I don't really know why that would happen.

So, could you please provide the logs (after checking they dont contain any private info) ?

Thanks

sarf2k4, 2022-04-14:
Hi,
I was trying to print a small parts, a small gear with 3mm hole in it. The printer intermittently paused on any very small movement gcode package like 3mm hole or the teeth of the gear. That would lead me to assume that the pi3b+ also had fully utilize its cpu resources.

During the intermittent pause, I did check the resource by using your plugin, to my surprise, cpu resource still has more room for others. The utilization were around 20-40% if i recall correctly, memory were around 15-30% (couldn't remember). However, as my memory serves me correctly, from the `resource plugin` tab it shows at least 25-35% extra resources left for cpu and ram

At first, I never suspected this plugin were the root cause of this, hence I tried to clean up several plugins that i don't really use during real printing because they really had no purpose to my use case either. I also gotten rid of the `octoprint-dashboard` where I thought it would be quite heavy on the pi hardware resource too, the pauses still continues. 

I remembered that this plugin has `1 sec` interval updates and I thought that would be the root cause, so I disabled this, printed the same object, there were no pauses anymore. I even re-installed back some of those gui plugin types such as:
- terminal messaging
- octoprint dashboard
- prusa m73 eta 
- and a bit of others I couldn't remember

Then I printed another parts that should have some micro movement gcode command object without the `resource monitor` plugin enabled, the results were great, the print job had zero pauses on the 3mm hole or small gap zig-zag solid infills.

Forgive me for not specifying my exact hardware other than the pi itself.
I am using prusa i3 mk3s mmu2s

sarf2k4, 2022-04-15:
Here are some of the logs as you've requested, I tried to 'detect' any pauses in the serial log and most of them has only 10-20ms interval between each other, while the pauses, is 1 second.

octoprint log however, I set this plugin to be on debug level. I searched for 'error' in the octoprint log in this reply, i found the following `	Line 1414: 2022-04-15 08:09:53,075 - octoprint.plugins.resource_monitor - DEBUG - disk_partitions() : [sdiskpart(device='/dev/root', mountpoint='/', fstype='ext4', opts='rw,noatime', maxfile=255, maxpath=4096), sdiskpart(device='/dev/mmcblk0p1', mountpoint='/boot', fstype='vfat', opts='rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro', maxfile=1530, maxpath=4096)]
`

Please ignore anything about `communication timeout` or `resend xx` lines as it is a known issues that is not related to this current plugin
[octoprint.txt](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/8492987/octoprint.txt)
[serial.txt](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/8492989/serial.txt)

jsdokos, 2022-04-17:
I have also run into the same issue. I narrowed it down to just this plugin. I made a small shape to test after removing a plugin. I removed all but this one and still had the same issue. Once i removed the plugin or started in safe mode the issue was no longer present. I have my system bundle and a video of the issue (not the same print).

https://www.youtube.com/watch?v=AScdG_whouw

[Systembundle viewer here: ](https://bundleviewer.octoprint.org) [octoprint-systeminfo-20220417033548.zip](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/8500890/octoprint-systeminfo-20220417033548.zip)

sarf2k4, 2022-04-17:
> I have also run into the same issue. I narrowed it down to just this plugin. I made a small shape to test after removing a plugin. I removed all but this one and still had the same issue. Once i [...]

Thank you, this has confirmed that this plugin hampers the pi performance a lot even if the resource says otherwise.

Years back perhaps 4 years ago I used pi0w on my printer, if I remember correctly I also had this installed back then, I immediately ruled out that pi0w is severely underpowered therefore not suitable to be used as octopi server.

I believe this plugin also need to be updated considering that octoprint and linux is starting to make pithon 2 obsolete

Renaud11232, 2022-04-17:
Hello,

I'm still trying to figure out what the issue might be. But not being able to reproduce that behavior on my printer makes it quite hard.

@jsdokos, could you provide the gcode of that test shape you tried to print ? I'll see if printing the same file does show the issue

Thanks

sarf2k4, 2022-04-17:
@Renaud11232 
It also depends on what type of pi you're using, you won't be able to reproduce if you're using pi4b, I think its easier to reproduce if you're using pi0w (not the pi02w), or pi1 or pi2. I for sure using a usb connection cable, didn't use the gpio, however the pi0w years ago was through gpio. Not sure if underclocking your pi would be able to simulate this behavior but it may increase the chances of reproducing this issues.

I think its way easier if you use fuzzy skin feature enabled available on prusaslicer and cura, make sure you're using rounded objects that's roughly 5mm or less. Don't use dryrun plugin or any similar method where you will exclude any g-code regarding extruder and hotend related as well.

What about the error I mentioned found in the log file?

jsdokos, 2022-04-17:
@Renaud11232 I will grab the stl and gcode after some activities with family. It was just a basic cylinder that had 2 walls since I noticed a that seemed to do it the most. I was also using a pi 3b rev 1.2. 

I wonder if refactoring to python 3 would fix the issue. I know that octoprint is removing support for python 2 after this version. Another thought is that it could be the process is taking too long with some thing deprecated now, or something with buster or bullseye.

Renaud11232, 2022-04-17:
I use an OrangePi Zero (which should be slightly slower than the RPi 3) and don't have any spare RPi to try :/

The error you mentionned earlier: 

```
2022-04-15 08:09:53,075 - octoprint.plugins.resource_monitor - DEBUG - disk_partitions() : [sdiskpart(device='/dev/root', mountpoint='/', fstype='ext4', opts='rw,noatime', maxfile=255, maxpath=4096), sdiskpart(device='/dev/mmcblk0p1', mountpoint='/boot', fstype='vfat', opts='rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro', maxfile=1530, maxpath=4096)]
```
is actually not an error, it's just the output of the function listing disk partitions which include the file system options (in this case, remount the file system as readonly if any error occurs)

This plugin is already compatible with python 3, so I'm not too sure what you meant.

sarf2k4, 2022-04-18:
Not a direct comparison between broadcom vs allwinner chip even if its the same architecture. What about underclocking your orangepi?

How about the amount of command queries this plugin will make given with that set interval lets say 1 second interval settings?
Perhaps you can reduce the amount of command queries to bare minimum during printing. The printers now will state its status whether it is operational, printing or idle

Renaud11232, 2022-04-18:
Hello,

I've tried lowering the CPU frequency of my OrangePi as low as 300MHz and still haven't got the issue even while running `stress --cpu 4` in the background to create a load on the CPU.

However, after looking again at the logs you both provided, I think I noticed a pattern : 

```
2022-04-17 03:34:23,084 - octoprint.plugins.resource_monitor - DEBUG - disk_usage(partition["mountpoint"] : sdiskusage(total=264289280, used=50511360, free=213777920, percent=19.1)
2022-04-17 03:34:23,092 - octoprint.plugins.resource_monitor - DEBUG - net_io_counters(pernic=True) : {'lo': snetio(bytes_sent=8254966181, bytes_recv=8254966181, packets_sent=1566153, packets_recv=1566153, errin=0, errout=0, dropin=0, dropout=0), 'eth0': snetio(bytes_sent=0, bytes_recv=0, packets_sent=0, packets_recv=0, errin=0, errout=0, dropin=0, dropout=0), 'wlan0': snetio(bytes_sent=0, bytes_recv=0, packets_sent=0, packets_recv=0, errin=0, errout=0, dropin=0, dropout=0), 'wlan1': snetio(bytes_sent=8661301154, bytes_recv=307487609, packets_sent=6672247, packets_recv=3084668, errin=0, errout=0, dropin=0, dropout=0)}
2022-04-17 03:34:23,096 - octoprint.plugins.resource_monitor - DEBUG - net_if_addrs() : {'lo': [snicaddr(family=<AddressFamily.AF_INET: 2>, address='127.0.0.1', netmask='255.0.0.0', broadcast=None, ptp=None), snicaddr(family=<AddressFamily.AF_INET6: 10>, address='::1', netmask='ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff', broadcast=None, ptp=None), snicaddr(family=<AddressFamily.AF_PACKET: 17>, address='00:00:00:00:00:00', netmask=None, broadcast=None, ptp=None)], 'wlan1': [snicaddr(family=<AddressFamily.AF_INET: 2>, address='192.168.50.250', netmask='255.255.255.0', broadcast='192.168.50.255', ptp=None), snicaddr(family=<AddressFamily.AF_INET6: 10>, address='fe80::a962:ec0c:2a2b:3641%wlan1', netmask='ffff:ffff:ffff:ffff::', broadcast=None, ptp=None), snicaddr(family=<AddressFamily.AF_PACKET: 17>, address='ac:22:0b:96:be:48', netmask=None, broadcast='ff:ff:ff:ff:ff:ff', ptp=None)], 'eth0': [snicaddr(family=<AddressFamily.AF_PACKET: 17>, address='b8:27:eb:84:99:af', netmask=None, broadcast='ff:ff:ff:ff:ff:ff', ptp=None)], 'wlan0': [snicaddr(family=<AddressFamily.AF_PACKET: 17>, address='b8:27:eb:d1:cc:fa', netmask=None, broadcast='ff:ff:ff:ff:ff:ff', ptp=None)]}
2022-04-17 03:34:23,954 - octoprint.plugins.resource_monitor - DEBUG - net_if_stats() : {'lo': snicstats(isup=True, duplex=<NicDuplex.NIC_DUPLEX_UNKNOWN: 0>, speed=0, mtu=65536), 'eth0': snicstats(isup=False, duplex=<NicDuplex.NIC_DUPLEX_UNKNOWN: 0>, speed=65535, mtu=1500), 'wlan0': snicstats(isup=False, duplex=<NicDuplex.NIC_DUPLEX_UNKNOWN: 0>, speed=0, mtu=1500), 'wlan1': snicstats(isup=True, duplex=<NicDuplex.NIC_DUPLEX_UNKNOWN: 0>, speed=0, mtu=1500)}
2022-04-17 03:34:23,956 - octoprint.plugins.resource_monitor - DEBUG - sensors_battery() : None
2022-04-17 03:34:24,003 - octoprint.plugins.resource_monitor - DEBUG - cpu_freq() : scpufreq(current=700.0, min=600.0, max=1200.0)
2022-04-17 03:34:24,008 - octoprint.plugins.resource_monitor - DEBUG - cpu_percent(percpu=True) : [7.2, 0.0, 3.1, 8.0]
```
I don't know why yet, but the call to `net_if_stats()` seem to hang sometimes, in this case for nearly 900ms.

Currently, even when the `Graph refresh rate` is set to more than 1s, the plugin actually keeps checking for resource usage every second and then updates the graphs with the average utilization computed over the configured period. This was requested a few years ago and I don't really want to remove a requested feature.

Would it be possible to have logs of the plugin when the printer isn't printing ? I'd like to see if this hanging/delay with `net_if_stats` also happens when the printer is not busy.

Thanks for your help and patience

sarf2k4, 2022-04-18:
what does the said command does?

I believe my logs also has some of the log before and after the intermittent pause print job, you just have to match the timing with serial log when it started printing

Renaud11232, 2022-04-18:
`net_if_stats()` returns information about the installed network cards, for instance the link speed (100Mb/s, 1Gb/s) that is then displayed under the graph.

While adding an option to increase the interval between queries would probably work arround the issue., the core problem wouldn't be fixed. This function call really should not take this long. (assuming that really is the issue)

sarf2k4, 2022-04-18:
What about disabling these queries while the printer is in `printing` state? 

Surely queries for disk capacity wouldn't be needed that much either, retain cpu and ram queries even while printing?

If this is a core issue that perhaps you couldn't solve by yourself, in which the `net_if_stats()` is core linux command, it may have to do with the kernel, outside the scope of this plugin,

However I believe you should try asking in octopi forum about this issues of `net_if_stats()` halting the pi's operation. This way you can confirm what needs to be done. Just me and @jsdokos won't help much as we are no expert in linux nor programming

jsdokos, 2022-04-18:
Here is my log for the printer not printing, I don't think I see an issue with it hanging, but i could be wrong. [octoprint.log](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/8506334/octoprint.log)

weeurey, 2022-05-21:
Same issue here! Running the plugin results in "stuttering" and blobs where the printing stops. Removed the plugin and printing is resuming as normal

Siggy101, 2022-05-21:
Hi all, 
I have the same issue with this plugin on my RPi 3B+ and Ender 3v2.
If it help, here is the gcode of a little hook I was printing when the stuttering started.  I also have some video and photos:  
[cable-holder-v3-remix OctoRebuilt.aw.gcode.txt](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/8748270/cable-holder-v3-remix.OctoRebuilt.aw.gcode.txt)
https://photos.app.goo.gl/gsDDADTGohRFuBx77

Monkeyware, 2022-05-24:
HI! Same issue here.
Rasberry 3B+, octoprint 1.8.0, Anycubis Mega S with Knutwurst-FM.
If plugin is enabled, the printer stops movement every second or so. Disabling this plugin, removes the pauses.

Renaud11232, 2022-05-27:
Hello,

I managed to get a RPi 3B+
After installing OctoPi on the SD card and setting it up, I still could not get the pauses to appear.

I'm trying to see what the common factor might be. So, it would be very helpful is all/most participants of this thread could answer these questions : 

* How is your RPi connected to the network ? (builtin ethernet/wifi, external wifi dongle/ethernet adapter/ other)
* Are there other devices connected to the RPi via USB or GPIO (in addition to the printer itself ? (network dongle, webcam, keyboard, anything, sensors, leds, ...)
* If yes, are these devices active/used when the problem occurs ?
* Are you using a USB hub ?
* Was octoprint installed manually ? Or using the OctoPi image ? Or any other image ?
* Are there any other pieces of software that you installed on the RPi in addition to octoprint ?
* Were you using Resource Monitor before the problem started appearing ? Or was it the first time you tried it out and it didn't work out of the box ?
* If it used to work properly, do you remember when the problem started appearing ? (after updating/changing something/...)

Thanks in advance

I really hope this will help me find the cause of the issue.

sarf2k4, 2022-05-27:
Mine using wifi, no other device than the printer itself connected to the rpi via usb only. I think my octopi were from an old image somewhere 1.6.x? that has been years so I just upgraded the octopi itself whenever there’s an update to it. As when these stuttering started to be so apparent, I think this issue started a while back like after around 6 months of pi0w came out, I fitted it into my printer, I just flashed a new image into it with this plugin installed. At first there were no problems until I had to print a small geometries like an m3 hole where there are tonnes of micro movement generated by the slicer; the stuttering were so apparent. However I quickly summarised that it’s the pi0w’s cpu not being powerful enough to handle those micro gcode movement. Which I described in my previous post above. So this issue started to surface again even on rpi3b+, however, you may want to try out on octoprint 1.7.x because 1.8.x just been released like months after this issue were opened. V1.8.x may have addressed this issue. As for my printer, I am using prusa mk3s, based on marlinfw, not sure about others who has the stuttering issues using which firmware. From: Renaud11232Sent: Friday, 27 May, 2022 9:20 PMTo: Renaud11232/OctoPrint-Resource-MonitorCc: sarf2k4; AuthorSubject: Re: [Renaud11232/OctoPrint-Resource-Monitor] [BUG] Print Job Intermittently Paused Then Continues (Issue #37) Hello,I managed to get a RPi 3B+After installing OctoPi on the SD card and setting it up, I still could not get the pauses to appear.I'm trying to see what the common factor might be. So, it would be very helpful is all/most participants of this thread could answer these questions :How is your RPi connected to the network ? (builtin ethernet/wifi, external wifi dongle/ethernet adapter/ other)Are there other devices connected to the RPi via USB or GPIO (in addition to the printer itself ? (network dongle, webcam, keyboard, anything, sensors, leds, ...)If yes, are these devices active/used when the problem occurs ?Are you using a USB hub ?Was octoprint installed manually ? Or using the OctoPi image ? Or any other image ?Are there any other pieces of software that you installed on the RPi in addition to octoprint ?Were you using Resource Monitor before the problem started appearing ? Or was it the first time you tried it out and it didn't work out of the box ?If it used to work properly, do you remember when the problem started appearing ? (after updating/changing something/...)Thanks in advanceI really hope this will help me find the cause of the issue.

Siggy101, 2022-05-27:
How is your RPi connected to the network ? (builtin ethernet/wifi, external wifi dongle/ethernet adapter/ other)
- Built in wifi
Are there other devices connected to the RPi via USB or GPIO (in addition to the printer itself ? (network dongle, webcam, keyboard, anything, sensors, leds, ...)
- 1 x Microsoft lifecam webcam
If yes, are these devices active/used when the problem occurs ?
- Yes, always active
Are you using a USB hub ?
- No
Was octoprint installed manually ? Or using the OctoPi image ? Or any other image ?
- Standard image install
Are there any other pieces of software that you installed on the RPi in addition to octoprint ?
- No, it only rund Octoprint
Were you using Resource Monitor before the problem started appearing ? Or was it the first time you tried it out and it didn't work out of the box ?
- I just started using it (after using OctoPi for a couple of months with no stuttering) and it started terrible stuttering immediately.  As soon as I disabled this plugin, it returned to normal
If it used to work properly, do you remember when the problem started appearing ? (after updating/changing something/...)
- N/A

Trashwarez, 2022-05-27:
In can confirm these issues on my Raspberry 3B+, Octoprint 1.8.1, Raspbian Buster, Artillery Genius.
When the plugin is enabled, the printer stops movement each 2 - 3 seconds, mainly on curves. Disabling this plugin, removes the pauses.

My system was installed from an OctoPi image. It connects via Wifi and uses an USB webcam and an USB stick, both directly connected to the Pi (no USB hub).

It worked fine since more than two years. As I am doing backups, I was able to find out that this issue exists at least since March 22. My backups dated Nov. 12, 2021 and before work fine with no hick-ups at all.

oakbrad, 2022-05-29:
Same issue. Raspberry Pi 3B+, Octoprint 1.8.1 (Octopi image), Ender 3 printer

> * How is your RPi connected to the network ? (builtin ethernet/wifi, external wifi dongle/ethernet adapter/ other)
Built in Wifi

> * Are there other devices connected to the RPi via USB or GPIO (in addition to the printer itself ? (network dongle, webcam, keyboard, anything, sensors, leds, ...)

Test with and without USB webcam attached, same issue. Nothing else attached, no USB hub.

> * Was octoprint installed manually ? Or using the OctoPi image ? Or any other image ?
Octopi
> * Are there any other pieces of software that you installed on the RPi in addition to octoprint ?
No just Octopi
> * Were you using Resource Monitor before the problem started appearing ? Or was it the first time you tried it out and it didn't work out of the box ?
I immediately started having issues after installing resource monitor, otherwise printer ran perfectly. Disable plugin, goes back to printing normally.

I'm using Ender 3 and it happens on curves - even simple shapes or small gcode files.

I have multiple printers I will see if I can reproduce on the others when they're not printing.

XXX-ManZG, 2022-06-19:
I have same problems as above with plugin

qwake22, 2022-06-23:
Same problem. Narrowed down to this plugging l. As a total newb it took a long while to figure this out.

Raspberry Pi 3b+
Ender 3 S1 Pro
Any of the retraction or stringing tests with thin posts were able to reproduce.

Trashwarez, 2022-06-30:
Not much action here to solve that issue - Uninstalled.

mriscoc, 2022-07-14:
Confirmed: https://github.com/Renaud11232/OctoPrint-Resource-Monitor/issues/39
Possibly the bug started after sudo apt update/upgrade.

DerbyOli, 2022-07-15:
Hello,

I can also confirm this problem.

I use four Raspberrys, all of them being Pi 3B or 3B+.

All four have been updated with apt-upgrade within the last few weeks, and all have exactly the same configuration (except of the attached printer type). The problem with the pausing and causing blobs occurs only on ONE of them. The attached printer is a Sovol SV04 (IDEX type) printer.

The other three Pi's are attached to 2x Anycubic Mega S and 1x Sovol SV02. They do NOT show this behaviour, despite also being upgraded. 

Hope this helps!

Greetings
Oliver

Renaud11232, 2022-07-18:
Hello,

I've made some changes to the plugin to try to get some more information on this issue.
Could you please install this version : https://github.com/Renaud11232/OctoPrint-Resource-Monitor/archive/refs/heads/perf.zip ? This is not available as a plugin update and needs to be installed using the URL or by uploading the zip file in the plugin manager.

Once installed, you may get a notification for an update. Ignore it for now as it will revert back to the "normal" version of the plugin. Instead, go to the plugin settings. In the "Advanced options", click on "Run a performance test" and link the downloaded file here.

[image]

While you are in the settings, check the "Enable profiling" option and click save. While this option is enabled, try to reproduce the pausing issue.
Once that's done, go back in the settings and disable the "Enable profiling" option. Do not let this option enabled as it will create very large log files over time.
Once disabled, go in In the OctoPrint settings, on the "Logging" page, download the "plugin_resource_monitor_profiling.log" log file and link it in this issue as well.

Thanks a lot and sorry again for the inconvinience

DerbyOli, 2022-07-20:
Hello Renaud,

I did the tests you requested and will attach the files here. I started a print with logging enabled and stopped it after about 15 Minutes, as I hope this should create enough data for you. 

I also attached a photo where I marked the blobs which are caused by the pausing, just FYI.

[psutil_timings.txt](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/9148199/psutil_timings.txt)
[plugin_resource_monitor_profiling.log](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/9148200/plugin_resource_monitor_profiling.log)
[image]

fudge01010, 2022-08-19:
Just chiming in to say that I have the same issue! Only on prints with lots of corners or small lines.
Never had the issue previously, had used this plugin without issues. Did a reimage to bring my base octopi image up to date (and then restored from backup) and started encountering this issue.

- How is your RPi connected to the network ? (builtin ethernet/wifi, external wifi dongle/ethernet adapter/ other)
  - wifi
- Are there other devices connected to the RPi via USB or GPIO (in addition to the printer itself ? (network dongle, webcam, keyboard, anything, sensors, leds, ...)
  - pi camera via the camera FFC
- If yes, are these devices active/used when the problem occurs ?
  - yes, always active
- Are you using a USB hub ?
  - no
- Was octoprint installed manually ? Or using the OctoPi image ? Or any other image ?
  - image
- Are there any other pieces of software that you installed on the RPi in addition to octoprint ?
  - just octoprint with a few other plugins. Stuttering happens with or without other plugins
- Were you using Resource Monitor before the problem started appearing ? Or was it the first time you tried it out and it didn't work out of the box ?
  - had used previously without issues. installed fresh image and restored from backup and problem appeared.

Happy to do some troubleshooting if you need, dev :)

DirtMonger, 2022-08-29:
Just adding my 2 cents as I encountered this issue too, only when there are a succession of rapid movements.  My first case was in building a new raspi3B+ for a new printer.  I took an octoprint backup of an existing printer and restored on the new copy of octoprint.  Immediately I had the pausing issue, where it was not occurring on the printer I took the backup from.  Being that I had 15+ plugins, and didn't want to disable them one by one, I then made another new install of octoprint and manually added a few select plugins, resource monitor was not one of them.  The pausing issue was gone.  As the weeks passed I wanted some of the plugins I had skipped, one day I added resource monitor, and the pausing returned.  Being the only plugin I had added that day, I disabled RM, and the pausing stopped.  

The curious thing is I have RM running on two other printers, all on raspi3B+ models, and those other printers do not have the issue. 

Versions installed: 
Printer with issue: RM 0.3.6, Octoprint 1.8.1, OctoPi 0.17
Printer 1 without issue: RM 0.3.6, Octoprint 1.8.1, OctoPi 0.18
Printer 2 without issue: 0.2.7, Octoprint 1.8.1, OctoPi 0.17

How is your RPi connected to the network ? built in wifi
Are there other devices connected to the RPi ? Nothing additional
Are you using a USB hub ?  No
Was octoprint installed manually ?  Octopi Image for all
Other pieces of software?  No
Were you using Resource Monitor before ?  Using RM for a long time on other printers, but new install for new printer that had the issue.

Good luck with the issue.  Not being able to reproduce it makes it a lot tougher.

Renaud11232, 2022-09-06:
Hello,

Just to give a small update on this issue.
From what I saw in the different log files I got, it looks like `psutil` (the library this plugin uses to gather resource stats) sometimes takes up to 4 seconds to return results.
More precisely, when getting the network statistics and the child processes spawned by OctoPrint (used to compute the OP CPU usage).

I've opened an issue on the psutil repository.

rcmz, 2022-09-18:
Hello,

I had the same bug on my Raspberry Pi 3B+, and it took me a bit of time to troubleshoot that the issue was coming from this plugin. I think this plugin is very cool and I'm gonna miss it, but pausing randomly and causing blobs is a big issue ! So I think that this plugin should be taken out of the plugin repository while this bug is sorted out, or at least display a big warning message when people install/activate the plugin. What do you think ?

Renaud11232, 2022-09-18:
Hello,

I've published a new update.
This version adds a warning notification regarding this issue.

Siggy101, 2022-09-18:
> Hello,
> 
> I've published a new update. This version adds a warning notification regarding this issue.

This is the next best thing, after a fix.
Thanks for your efforts.

rcmz, 2022-09-18:
> Hello,
> 
> I've published a new update. This version adds a warning notification regarding this issue.

Thanks, that was quick !

Tr33x0rs, 2022-09-22:
> Hello,
> 
> I've published a new update. This version adds a warning notification regarding this issue.

Is there a way I can disable this message? I'm constantly tinkering with my Octoprint setup and it gets old having to close that message every time I login.

Renaud11232, 2022-09-23:
> Is there a way I can disable this message? I'm constantly tinkering with my Octoprint setup and it gets old having to close that message every time I login.

It should be possible starting with 0.3.8 (released just now). You need to go in the plugin settings > Advanced options

FredGenius, 2022-09-23:
I have been using this plugin for long time, with latest octoprint and octopi updates, I often print small, intricate things but have never had a problem as described by others here. This leads me to conclude that there must be another factor involved - maybe printer firmware? I'm using Marlin 2.0.9.1.

Renaud11232, 2022-10-02:
Hello,

I have made some changes on the `dev` branch : I've removed the calls to the `net_if_stats` function, which is the one that might be causing long delays.
Can you try to install the dev version of the plugin to see if the pausing issue is still present ? https://github.com/Renaud11232/OctoPrint-Resource-Monitor/archive/refs/heads/dev.zip 
This means that the speed of the network interface wont be displayed anymore.
If that fixes the issue, I'll add a configuration option to enable/disable this.

Thanks

nicolem, 2022-10-03:
I came to this bug from the new alert.  I've been having blobs on prints from octoprint, but not on the sd card.  I found that using an ethernet cable instead of wifi fixed the blob/pause issue for me. I went ahead and tried the dev branch you just posted, and it works fine on wifi. 

thanks so much for addressing this issue, because I had just started working through troubleshooting the blob/pause issue. 

to recap:
| branch      | internet | blobs? |
|-------------|----------|--------|
| master      | wifi     | yes    |
| master      | ethernet | no     |
| uninstalled | wifi     | no     |
| dev         | wifi     | no     |

don-willingham, 2022-10-04:
I flashed an old octopi-buster-armhf-lite-0.18.0.zip image with Octoprint 1.5.2,  Python 3.7.3, and a 5.4.79 kernel dated Nov 23, 2020. I can update that to 1.8.4, restore a backup of my current configuration and prints do not pause. Performing a apt update/upgrade causes the pause to return. Flashing a more recent octopi-0.18.0-1.8.4-20220927151125.zip with 5.10.103 kernel dated Mar 8, 2022 immediately pauses. Unfortunately octopi "0.18.0" seems to be a moving target. Resource Monitor (0.3.9rc1) fixes my pauses. It seems to be a problem in the underlying  image that octopi is built on. I'll experiment with old nightly builds from https://unofficialpi.org/Distros/OctoPi/nightly/ and see where it breaks. I may also try to selectively apt upgrade certain package(s). I am on wifi, and a Creality Ender 3, and some Systeminfo data:
```
env.hardware.cores: 4
env.hardware.freq: 1200
env.hardware.ram: 914006016
env.plugins.pi_support.model: Raspberry Pi 3 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
printer.firmware: Marlin TH3D UFW 2.26 (Apr 16 2021 21:08:21)
```

Danafra, 2022-10-04:
I'll post again later when I'm home with my specifics, but I've noticed this issue as well. At least I think I'm seeing this issue. The printer will pause briefly while printing. However, if I stop monitoring it with my computer (by either closing the browser window or shutting off the computer), the problem stops. I've just got used to doing that.

BTW, this is also with a Pi 3B+.

don-willingham, 2022-10-05:
2021-12-16_2021-10-30-octopi-bullseye-armhf-lite-1.0.0.zip does not pause (5.10.63 kernel)
2022-05-01_2022-04-04-octopi-bullseye-armhf-lite-1.0.0.zip pauses (couldn't ssh in to get kernel version)
2022-03-12_2022-01-28-octopi-bullseye-armhf-lite-1.0.0.zip pauses (5.10.92 kernel)

I will continue to try nightly builds between 2021-12-16 and 2022-03-12. These later nightly builds seem to have python 3.9.2 instead of 3.7.2 from my previous buster release.

don-willingham, 2022-10-05:
2022-01-18_2021-10-30-octopi-bullseye-armhf-lite-1.0.0.zip does not pause (5.10.63 kernel)
I suspect the next nightly, 2022-02-01_2022-01-28-octopi-default-bullseye-armhf-lite-1.0.0.zip, will have the problem, as it updates to 2022-01-28, the second bullseye release, like 2022-03-12_2022-01-28-octopi-bullseye-armhf-lite-1.0.0.zip; see https://en.wikipedia.org/wiki/Raspberry_Pi_OS

sarf2k4, 2022-10-05:
Anyone ever tried on pi0w? Not the pi0w2.

Perhaps using a pi0w or perhaps, pi1 (because similar in hardware performance), may amplify this problem which I described at https://github.com/Renaud11232/OctoPrint-Resource-Monitor/issues/37#issuecomment-1100861375

don-willingham, 2022-10-06:
2022-01-18_2021-10-30-octopi-bullseye-armhf-lite-1.0.0.zip does not pause. It has a 5.10.63 kernel. I did and `apt-get update` and `apt-get upgrade raspberrypi-kernel` which updated to 1.20220830 aka 5.15.61, which does pause when printing. I am trying 2022-02-01_2022-01-28-octopi-default-bullseye-armhf-lite-1.0.0.zip next.

don-willingham, 2022-10-06:
2022-02-01_2022-01-28-octopi-default-bullseye-armhf-lite-1.0.0.zip has raspberrypi-kernel 20220120-1, aka 5.10.92, and it pauses. So I'm assuming there's a problem introduced somewhere between the 5.10.63 and 5.10.92 kernels.

don-willingham, 2022-10-07:
Flashed 2022-01-18_2021-10-30-octopi-bullseye-armhf-lite-1.0.0.zip again. After updating to 1.8.4 and restoring my previous configuration, I ran:
```
sudo apt-get update
sudo apt-mark hold raspberrypi-kernel
sudo apt-get upgrade
```
So everything but the kernel would be upgraded. The upgrade reported:
```
The following packages have been kept back:
  libcamera-apps-lite libcamera0 raspberrypi-kernel raspberrypi-sys-mods
```
After a reboot, and test print, my printer doesn't pause, so I don't think any of the other packages are to blame but the kernel.

sarf2k4, 2022-10-07:
@Renaud11232 , can you confirm your raspberry pi's kernel version as what has been investigated by @don-willingham at https://github.com/Renaud11232/OctoPrint-Resource-Monitor/issues/37#issuecomment-1269217941 as what you stated that you're unable to reproduce such issues https://github.com/Renaud11232/OctoPrint-Resource-Monitor/issues/37#issuecomment-1100877692

Renaud11232, 2022-10-07:
Hello,

Thanks for investigating this.
I don't use a Raspberry Pi on my own 3D printer but an Orange Pi Zero running armbian (kernel `5.4.0-126`)
Unfortunately, I cannot check the kernel version of the RPi I used for testing anymore.
However, if this actually turns out to be an issue with the kernel version, I won't be able to do anything to fix that.
The best I can do is to stop calling the function that is affected by the kernel version changes.
2 people confirmed that the changes made on the dev branch (https://github.com/Renaud11232/OctoPrint-Resource-Monitor/archive/refs/heads/dev.zip) solved the issue. I'll try to release this in the near future.

The only side effect of this *fix* is that the speed of the network interface (100Mb/s, 1Gb/s, ...) won't be displayed anymore in Resource Monitor (by default). For the users that weren't affected by this issue and still want to have this information displayed, I'll add a configuration option that re-enables it.

I don't think I can do anything else.

closed 2022-10-10 by Renaud11232

Renaud11232, 2022-10-10:
Version 0.3.9 (released just now) contains the changes previously mentionned and should fix the pausing/freezing issue.

Thanks for your help with this issue

don-willingham, 2022-10-11:
Thanks @Renaud11232 👍
The release works, and if I enable `Display network interface speed`, the freeze returns. Since I'm on wifi, I tried to add Network exceptions for "eth0", then "eth0 and lo", and finally "eth0, lo, and wlan0"; the pauses come back in all 3 cases. It didn't occur to me until today to try that.
Unfortunately according to https://psutil.readthedocs.io/en/latest/, `psutil.net_if_stats()` provides: isup, duplex, speed, mtu, and flags. If the implementation of "speed" is this plugin cares about, but one of the others is causing the slow down, it might be worthwhile requesting the addition of an argument to specify what. It also enumerates all network interfaces, if this plugin could request for specific interfaces, based on the exception list; it might also avoid the pausing. (or submitting a PR or two to add that flexibility) Thoughts?

don-willingham, 2022-10-11:
I now see the discussion in https://github.com/giampaolo/psutil/issues/2133 already. If the problem is in `psutil._psutil_linux.net_if_duplex_speed`, then asking only for duplex and speed isn't going to help.

00ihsan, 2023-04-17:
Any update? I enabled it today and got the message again

commit e836c1e5 (2022-10-08): prerelease 0.3.9rc1
removed performance test
adds option to re-enable it
fixes #37