Renaud11232/OctoPrint-Resource-Monitor #38: [BUG] This plugin causes stuttering

Siggy101, 2022-05-21:
**Describe the bug**

Installed on OctoPrint a RPi 3B+ running OctoPrint 1.8.0, this plugin causes the Ender 3v2 to stutter all through a print job.

**To Reproduce**

Steps to reproduce the behavior:
Print anything with this plugin enabled.
If I disable this plugin, and only this plugin, the printer runs smoothly again.

**Expected behavior**

Print smoothly without stuttering

**Version of OctoPrint-Resource-Monitor**

0.3.6 installed via the Octoprint plugin manager

Renaud11232, 2022-05-21:
Hello, I am aware of this issue affecting multiple users.
But I can't reproduce it on my own printer nor can't find what causes this.

I'm closing this issue since this is a duplicate of #37

closed 2022-05-21 by Renaud11232

Siggy101, 2022-05-21:
Ah, sorry, I missed that other bug report.
I shall follow the other then.  I can post up my gcode and a video, if it helps.