Renaud11232/OctoPrint-Resource-Monitor #40: [FR] can you please Add octo4a support to this plugin ?

j-snijder, 2022-10-30:
The feature I like to see is NOT related to a problem with octoprint as persé, but more as a problem loading in octo4a (which is a derivate of octoprint)
After installation there should be, but is not, a new tab where we can see the graphs with information.
  
I'd lik to see a compatibility with octoa, if you could be so nice :)

Please help out, as more and more people are discovering the use of octo4a in stead of the RP option for which people need to be a rocket sciëntist to get it up and running. :)

Renaud11232, 2022-10-31:
Hello,

I don't know much about octo4a more specifically how plugins are handled/installed.
Can you please provide octoprint logs ?

Thanks

j-snijder, 2022-10-31:
Hi

The logfile is completely empty so nothing is logged, sorry.

j-snijder, 2022-10-31:
[octoprint-logs (1).zip](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/9900546/octoprint-logs.1.zip)
What I can provide you with is the octoprint log itself, perhaps thats good enough ?

Renaud11232, 2022-10-31:
Hello,

Yes I meant octoprint logs. That's perfect.
This issue appears to actually be a bug that is not specific to octo4a.

I just published a release candidate that should fix the issue.
Can you try to install it manually using the following URL through the plugin manager (Settings > Plugin Manager > Get more > from URL) : https://github.com/Renaud11232/OctoPrint-Resource-Monitor/archive/refs/tags/0.3.10rc1.zip

If that didn't work, can you send your latest logs again ?

Thanks

j-snijder, 2022-10-31:
Hi 😁 as I always say, coincedense doe not exist, everything is ment to be 🤣I'll install it this afternoon and contact you asapKind regards Jan

j-snijder, 2022-10-31:
[octoprint-logs.zip](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/9902610/octoprint-logs.zip)
Hi,

Installed the beta version and sorry to say but still nothing to see, no new tab, no entry in the plugin listing.
As before there is a logfile created but empty as before (attached it also for you to see)

Are there some specific prerequests to this plugin ? is it neccesary to have some other plugins installed on what it is depending on ?

Like to help out in testing if needed to.

Kind regards, Jan

Renaud11232, 2022-11-11:
Hello,

It appears that the first issue was in fact hiding the next one which is specific to android.
I've just published a new prerelease that addresses that issue and should allow the plugin to run : https://github.com/Renaud11232/OctoPrint-Resource-Monitor/archive/refs/tags/0.3.10rc2.zip
I hope there isn't another issue down the line.
However, that also means that some (or all) partitions won't appear in the plugin tab. This is due to the fact that Android does not give read access to the files needed to get disk usage information. I don't see any way to work around that.

Can you tell me if that worked ?

j-snijder, 2022-11-13:
Hi,Sorry,stil not working, nothing shows upRegards, Jan

Renaud11232, 2022-11-13:
Can you provide more recent logs to see if any new issue showed up ?

j-snijder, 2022-11-13:
Hi,

Sorry it took so long, had to do a uninstall of the previous version and a fresh install of your last updated version.

Attached is the octoprint log
[octoprint.log](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/9998099/octoprint.log)

Renaud11232, 2022-11-15:
Hello,

Once again, this issue was hidden by the previous one. I hope there won't be anymore suprises.
Can you try to install this version : https://github.com/Renaud11232/OctoPrint-Resource-Monitor/archive/refs/tags/0.3.10rc3.zip and report if that fixed the problem ?

If not, please provide recent logs.
Thanks for your patience

j-snijder, 2022-11-15:
Hi :(

Sorry to say that it still does not show up, neither in the plugins list nor as a tab in octoprint :(

But... together we stand strong :)

[octoprint (1).log](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/10012037/octoprint.1.log)

Renaud11232, 2022-11-27:
Hello,

I've managed to successfully install octo4a on 2 devices and got the plugin to run on those two devices with a new prerelease. (https://github.com/Renaud11232/OctoPrint-Resource-Monitor/archive/refs/tags/0.3.10rc4.zip)

However, due to android file permissions being so restrictive, the plugin fails to gather a lot of resource usage information : 
* On the 1st test device, some disk usage info & some network usage could be gathered and displayed.
* While on the 2nd device, only RAM & CPU usage could be monitored (with some data looking abnormal).

This will probably be different on other devices and/or might even render the plugin completely useless.

Can you tell me if that version worked ? (It might be pretty empty due to android file permissions)
If not, can you once again provide up to date logs ?

Thanks a lot for your help

j-snijder, 2022-12-01:
Hi,
Sorry for my late reaction, had my birthday so got a lot of gifts I wanted to implement on my printer (mega upgrades :) )

Okay, installed the latest rc4 and it works on a lot of stuff, though the screen look very strange (i'll attach some screens for you to examine.

Perhaps its in the way I (did not yet) setup the app)

[image]
[image]
[image]

j-snijder, 2022-12-01:
Memory
Network 4
Network 15
Battery

All seem to work, on my Samsung A5 (2017)
[image]
[image]
[image]
[image]

j-snijder, 2022-12-01:
Processor seems to work and reports Cores, threads and processes but shows no graph 

Kind regards, maybe this helps you to solve the plugin not working on octo4a

Renaud11232, 2022-12-02:
Hello,

Can you please set the plugin logging to DEBUG and provide the logs once again.
To set it to DEBUG, you need to go in Octoprint settings -> Logging -> Logging Levels
[image]

Once you have sent the logs, you can remove that log configuration to avoid unnecessary logging.

Thanks

j-snijder, 2022-12-04:
I have deinstalled the plugin for now, something is causing big problems in octoprint resulting in heavy processor use and very bad printing as a result.Have to find out, after removing this plugin things got better but not fully back to normal.All started after installing the v3 of your fix 😁I'll get back to you, but if I forget and you haven't heard from me in a week please remind me.Regards

Renaud11232, 2022-12-17:
Hello,

Have you had the time to look into this ?

THanks

j-snijder, 2022-12-17:
I'll try tonight okay ? Grandchildren now in tha house 🤣🙈 disaster time,😂

j-snijder, 2022-12-17:
Hi,

I'm sorry to report that resource manager does not log , not one bit, the logfile: octoprint.plugins.resource_monitor,   is not shown in the list of logfiles, therefore not downloadable.
The file :  octoprint.plugins.resource_monitor.profiling, does show but is 0 bites.

[image]

Renaud11232, 2022-12-18:
When you set the debug level to `DEBUG`, the plugin does not create a new file but keeps logging in the `octoprint.log` with more details.
`octoprint.plugins.resource_monitor.profiling` should not contain anything as it's used only if you enable an option which is not required here.

Can you please provide the contents of octoprint.log when you have the logging level set to `DEBUG`.

Thanks & sorry if it wasn't clear.

j-snijder, 2022-12-18:
Hmmm, logfile too big, max is 25 Mb file is 28 Mb

Renaud11232, 2022-12-18:
You should be able to upload it here : https://www.dropbox.com/request/P5LetK5NfxccD1m7dKrb

j-snijder, 2022-12-18:
Hi, sorry but a new problem occured.

As I deleted the old logfiles now octoprint is no longer logging anythin anymore, even if I tell specificly to log certain files.

I'll upload my old log to your dropbox.

regards, Jan

j-snijder, 2022-12-18:
Done uploading to your dropbox.

regards, Jan

Renaud11232, 2022-12-26:
Hello,

I have not been able to reproduce the layout issue with the strange graph placement. Even after installing the same plugins as you. However I think it might be linked to one of the UI plugins you are using, maybe with a specific configuration.

I don't think I can't do much if the CPU graphs don't show anything as `psutil` (the library I'm using to get resource usage stats) does return 0 all the time in this case.

The only way to know which one is to disable them until everything works out normally.

I'm sorry if what I'm asking sounds tedious but can you try disabling your plugins one by one until the graphs show up normally ? And then tell me which one it was (maybe with its configuration) so I can see if I can do something to fix that ?

Once again, thank you

j-snijder, 2022-12-26:
Hi 😁As octo4a gives me lots of problems lately I have left the programm for now, if and when I have the time to see whats wrong I'll try your options and remove the plugins to test.That, will be somewhere end January at its earlyest, sorry to say.In the meantime I wish you a happy newyear and whats left of Christmas 😁Greets, Jan

bend1215, 2023-01-17:
Hey thanks for putting such a cool plugin together.

I'm also running Octo4a but on a Pixel 3. Using the main release I also was missing the added tab. Using the latest release above I got an empty CPU usage graph but a working RAM graph, no other readouts:
[image]

Seems as though its the same problem as the other user, not sure if you're looking for more logs, I think I've followed the settings you gave above. 
[octoprint.log](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/10439376/octoprint.log)

I'm primarily interested in the temperature monitoring as this has been the only reason the phone controlled printing has been unstable. It would be nice to be able to monitor its temperature and shut down the IP cam streams for example. 

Thanks again!

Renaud11232, 2023-01-21:
Hello @bend1215,

It looks like you set the logging level of `octoprint.plugins.resource_monitor.profiling` to `DEBUG`.
Can you try again with `octoprint.plugins.resource_monitor` (without `.profiling` at the end) instead and then upload newer logs here ?

That being said, I see that the plugin could not get some resource stats due to android file permissions, I don't think there's a way to prevent that behaviour. In this case a `PermissionError` occured when trying to get : 
* CPU frequency
* disk partitions usage
* network interface usage
* battery stats

Thank you for your understanding

Renaud11232, 2023-04-10:
Closing since I can't get arround android permissions

commit 7b6ad325 (2022-10-31): fix settings migration failure
fixes #40
closed 2023-04-10 by Renaud11232