Renaud11232/OctoPrint-Resource-Monitor #41: [BUG] Temperature not showing on Orangepi Zero2

7R0J4Ner, 2022-12-18:
**Describe the bug**

The temperature of the Orangepi is not displayed.

**Screenshots**

[image]

**Version of OctoPrint-Resource-Monitor**

0.3.9
Installed using pluginmanager

**Server configuration**
 - OS: Linux 4.9.170-sun50iw9
 - Browser Firefox 108.0.1 (64-Bit)

**OctoPrint logs with `octoprint.plugins.resource_monitor` set to DEBUG**

https://www.dropbox.com/t/qNAYccVjY2FCQxId

Renaud11232, 2022-12-18:
Hello,

I just released 0.3.10 which should now display OPi Zero 2 CPU temperature.

closed 2022-12-18 by Renaud11232

7R0J4Ner, 2022-12-18:
Wow, that was Fast.
Works perfectly.
Thanks a lot!

commit 4a162dd6 (2022-12-18): adds support for Orangepi Zero2 CPU temperature
fixes #41