Renaud11232/OctoPrint-Resource-Monitor #42: Resource Monitor Disappears when you setup / use CAN BUS

JesNoFear, 2023-02-12:
The plugin disappears from  Settings -> Plugins and from the tab after setting up CANBOOT
Still shows up in Plugin Manager.

I did uninstall it and reinstall it, thinking it may have gotten corrupted but no dice.

I was curious the amount of data being sent over CAN0, then noticed the tab was missing.

Renaud11232, 2023-02-12:
Hello,

Can you put the plugin logging level to `DEBUG` in the settings and upload them here.

Thanks

JesNoFear, 2023-02-14:
Do you have a way to change the log level via command line or editing a file, as the plugin no longer shows up in the plugins tab.

[image]

[image]

Renaud11232, 2023-02-18:
Hello,

You can set the logging level from OctoPrint's Logging settings page : If possible, you need to add a log level configuration for `octoprint.plugins.resource_monitor` (not the one ending with `.profiling`) with the `DEBUG` level.

[image]

Then, from the same page, you can download `octoprint.log` and upload it here.

Sorry if this wasn't clear

Thanks

JesNoFear, 2023-02-20:
[octoprint.log](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/10787876/octoprint.log)

it looks unhappy on line 1547

is it because it see's can0 as an interface without a address on it?

Renaud11232, 2023-02-21:
Hello,

The errors occurs because `psutil.net_if_addrs()` did not return any address for that interface.
However I had already released a prerelease that adresses this issue but did not publish it on the main branch.

Can you try to install version `0.3.12rc1` of the plugin using this URL : https://github.com/Renaud11232/OctoPrint-Resource-Monitor/archive/refs/tags/0.3.12rc1.zip
You need to go to the OctoPrint Settings > Plugin Manager > + Get more > from URL and use the URL above.

If that solves the issue, I'll release this.

Thanks

JesNoFear, 2023-02-23:
That worked.

[image]

Thank you

closed 2023-02-23 by JesNoFear

Renaud11232, 2023-02-23:
Hello,
I just released this into `0.3.12` as a normal (non prerelease) update.

Thanks for your help

commit ea186cd2 (2023-02-23): release 0.3.12
fixes #42