Renaud11232/OctoPrint-Resource-Monitor #43: [BUG]

Otaku-Labs, 2023-03-18:
**Describe the bug**
Running OctoPrint on a Dell Wyse 5010 Thin Client, there's not a cpu temp read-out. 
One solution would be to add '**acpitz**' to '**temperature_possible_keys**' _[line 18, monitor/\_\_init\_\_.py]_
The tc actually responds with acpitz, radeon, and k10temp; Which, I can only assume the latter are the integrated graphics and chipset, respectively.

**To Reproduce**
Niche audience and would require similar hardware.

**Expected behavior**
Display CPU temperature

**Screenshots**
-Current code (Not working)
[image]

-Working fix (adding key 'acpitz' to list)
[image]

**Version of OctoPrint-Resource-Monitor**
OctoPrint: 1.8.7
Resource Monitor: 0.3.12

**Server configuration**
 - OS: Ubuntu 22.04
 - Browser: Chrome 111.0.5563.65 (Official Build) (64-bit)
 - Python version: 3.10.6

Renaud11232, 2023-03-18:
Hello,

I've added `acpitz` to the list of possible keys and released it as a prerelease https://github.com/Renaud11232/OctoPrint-Resource-Monitor/archive/refs/tags/0.3.13rc1.zip

Can you try to install it and see if that works ?
If this works, I'll release it as 0.3.13

Thanks

closed 2023-03-25 by Renaud11232

Otaku-Labs, 2023-03-29:
Hey Renaud,

Sorry, I didn't notice you had replied. I had already changed the code on the local install and it was working. I say was because I kinda broke my network configs messing with some things yesterday on the linux side, trying to get wifi up. 

I'll get around to fixing it sometime this weekend, if not before then, and report back; but, I'm 99% certain it'll work.

Thanks

commit 5976e87f (2023-03-18): adds acpitz to possible temperature keys
fixes #43