Renaud11232/OctoPrint-Resource-Monitor #44: [BUG] webui hangup

EvilGremlin, 2023-04-06:
It seems Resource Monitor (0.3.13rc1) cause interface hangup in octo 1.9.0rc3 after >1day uptime. Everything else works, but webui become completely unresponsive.

**Server configuration**
Ubuntu 22.04.1

will add logs later when it'll hang again, gotta print for now
[plugin_list.txt](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/files/11169871/plugin_list.txt)

Renaud11232, 2023-04-10:
Hello,

I suspect the logs octoprint logs won't be containing much information. However, the browser logs might.
If this happens again, can you paste your browser logs here ? (Right click > Inspect > Console tab).

The only possible cause I see is that the browser tab where the OctoPrint page was open got suspended. That can happen if the tab is inactive in the background. Then, once the tab is reopened. all graphs are updated to get back to the present, which might result in a lot of computation and thus, hangup.

I'll see if there are ways to optimize the code so it does not hang (for that long)

EvilGremlin, 2023-04-10:
It was complete hangup, not just delayed. I waited for like 10 minutes or even more. And now that i enabled your plugin back it don't hang up because of course it doesn't...

Renaud11232, 2023-04-12:
I'll keep this issue open for a while in case it happens again.
If so please let me know.

(It will auto-close in 60 days)

closed 2023-06-20 by github-actions[bot]